polyend/TrackerBetaTesting #1052: [1.3.1] Tempo FX off doesn't appear to do anything/tempo FX documentation confusion

tumes, 2021-05-09:
### Steps to reproduce

1. Set a tempo FX (T)
2. In a subsequent step, set a toggle off FX (!-)

I would expect this to reset the pattern tempo to project tempo, but perhaps that's not a reasonable expectation? It seems like the solution is to set a subsequent step to project tempo to get things back on track? But since the scale and precision of the tempo FX is different from the project tempo scale and precision, that workaround is not quite optimal.

Additionally, the documentation for the tempo FX is a little confusing. It refers to the tempo FX as being a percentage, but when I monitor the clock externally, it seems like the FX actually sets BPM directly and is neither a percentage of project tempo nor a percentage relative to the last tempo set by FX. If it _were_ a percentage of project tempo, then it'd be pretty easy to reset tempo in a subsequent step instead of attempting to use the toggle FX.

Sorry, I'm kind of lumping two things together here. I guess the real take homes are:
1. It'd be nice if the off FX worked on tempo (or if it was explicitly called out as _not_ working).
2. There's a mismatch between the function and documentation of the tempo FX.

### Attachments and links

N/A

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Every time.

* Does this bug happen in previous firmware versions?
I just got my tracker so I'm not sure! I'll take a look when I have some spare time.

* Does this bug happen in every project?
Yes, as far as I can tell.

PITmast, 2021-05-09:
Hi

Yes, FX T is a percentage of Song tempo.
If you want to double fast of Song Tempo - put at least one T200.
If you want slow half down of Song tempo - put at least one T 50.
If you want to back to Song tempo - put at least T100.

Checked, it works.
Regards

tumes, 2021-05-10:
@PITmast Hmmm, that's not what I'm seeing. Here's a pattern starting with the song set to 300 BPM, transitioning to a T20 FX step (20% of 300 -> 60 BPM), then to a T100 FX step (which I'm assuming is running at 100 BPM, but is at least clearly not back to 300 BPM). https://youtu.be/oEwG9hnnzoA

PITmast, 2022-02-24:
Hello everyone

In Tracker v.1.5.0.pdf docu:
T – Tempo changes pattern tempo from 10% to 400%. Handy to create custom swings or grooves. Notice that MIDI Clock output is based on the current tempo. Synchronizing anything over MIDI while using Tempo step Fx will affect the MIDI sync. It supports the full stop command with the STP setting > jog-wheel max to the left.

Tested with 1.5.0. From my perspective FX value is not a percentage of current tempo. It's a absolute tempo value. And this is good I guess, much simpler and more usable.

Regards

closed 2022-03-19 by MortalJS