polyend/TrackerBetaTesting #1054: Accessing the decimal place in Tempo

tfspsound, 2021-05-12:
We can use Up and Down arrows and Jogwheel to adjust the tempo in whole numbers but we can only access the decimal point by tapping tempo, unless I missed something. Could we use Left and Right arrows to adjust this? It would be helpful when we need to match the exact tempo of our creation to that of the sample used. 
Eg to exactly match a samples tempo that is 95.25 I would have to set a tempo of 381 but in doing so I am effectively reducing the pattern length limit to from 128 steps to 32. Some tempos would still be unreachable but we could get closer.

Jacknametrouble, 2021-05-14:
Yes, you've missed the manual probably - "Change the BPM value in decimal values (0.1) with the Shift key pressed."
https://polyend.com/manuals/tracker-manual/#song-mode

closed 2021-05-14 by Jacknametrouble

tfspsound, 2021-05-15:
Sorry, I’m sure I tried this and it didn’t seem to work. I’ll try again. Cheers