polyend/TrackerBetaTesting #1056: Simple FM instrument

lanthias, 2021-05-12:
I would absolutely love even a basic FM synth.

### A new feature

- A simple 3 or 4 OP FM synth as an additional instrument
- I'd be happy with an OP or DX clone
- Sine waveform, though others would be a nice to have
- User definable waveforms (like RCM on SY77/SY99) would be killer, but presuming much harder
- Modulation capabilities not required

I feel like this would take the Tracker into a totally new territory to compete with things like the Digitone (even with a less rich FM implementation).

I'm presuming the blocker here isn't computational power and there seem to be a fair few mature libraries for the algos, so would love to know what could be done to have this added.

I saw the request for an OPL3 on #975, so let me know if this is a dup.

MrMolletta, 2021-05-12:
It would be a KILLER feature! 🔥🔥❤️

okkobroderson, 2021-05-13:
+1 but modulation capabilities absolutely required!

0x0zzz, 2021-05-13:
I feel it would be in good spirit of the Tracker if the OP sources were the already-present wavetable synthesizer.  Might be a total dream though. :)

closed 2021-05-14 by Jacknametrouble

lanthias, 2021-05-18:
Out of interest @Jacknametrouble, what does closing an issue mean on this repo? It would be good to have some context or even a tag (e.g. "Won't do this", "We are considering this") - else it's not especially encouraging for users to contribute ideas.

bignutemporium1, 2021-05-27:
Shame, this would be great.

Marc-3000, 2021-12-30:
A simple FM synth like in the Nanoloop tracker/app would be awesome ;) 

https://apps.apple.com/us/app/nanoloop/id322700286

I don't know how difficult this is but if it's possible it would
make the Polyend Tracker even more awesome.

Thank you in advance! You Polyend people Rock!! ;)