polyend/TrackerBetaTesting #1061: Gate Length shouldn't just go shorter..

gherat, 2021-05-17:
There have been several requests of slices to play through. A slice should act as a startpoint "play sample from here" and then continue playing insteead of stopping at the end of the slice. The gate length should be in [1/32th]  [1/16th]  [1/8th]  [1/1]  [2/1] [4/1] or even 25% 50% but also 400% where the slice plays BEYOND the endpoint of the slice. This is a really important option for making breakbeats. Please make this option, it will be so much more useable and musical that way..

ketumatek, 2021-05-17:
For me gate is not enough specific for drums sample you put 6 or under no sounds and you put 7 , 8 or more and I don't see any difference.
Normally gate use to cut the sample as you like and exactly where you want ...
So may be difference between one to another ( ex gate 5 and gate 6 ) must be shorter, as this we can use it as it should...

bambrose, 2021-05-17:
I'm glad you raised this. I thought it was me. I am experiencing exactly the same. It just doesn't work.
Values of 5 or so do but that's about it