polyend/TrackerBetaTesting #1063: [1.4.1 beta] Line in Mono Left recording not summing and off behavior

sunfallz, 2021-05-20:
Hi, 
Line in Mono left recording not summing to mono when monitoring output, still only hear Left side output. 
Not sure if this is meant to happen still, but we’re on the right track with options now.
Would love to be able to sample anything in mono with a Mono patch cable and monitor both left and right output summed. 

Also, strange behavior:
when in Mono Left mode, with source still running, click on the Config button from the Sample Recording page, 
and suddenly you hear your mono source in both left and right as if it was summed from mono. 
( BUT this would be great if it was normal)!!!! 

* How often does this bug happen? 
* Every time

* Does this bug happen in previous firmware versions? no this is beta 1.4.1

* Does this bug happen in every project?
Yes

tfspsound, 2021-05-20:
#1026 same issue

sunfallz, 2021-05-21:
> #1026 same issue

Have you experienced this strange behavior too:

...when in Mono Left mode, with source still running, click on the Config button from the Sample Recording page,
and suddenly you hear your mono source in both left and right as if it was summed from mono.
Exit the Config page and you’re back to only hearing the Left output again. Odd..

MortalJS, 2022-02-02:
#1026 duplicate

closed 2022-02-02 by MortalJS