polyend/TrackerBetaTesting #1065: [1.4.0 Beta 1] ARP FX Bugs and Improvements

clickysteve, 2021-05-21:
### Bug One: Arp FX Command Interruption Inconsistent

If you add in a different FX Command into either the FX1 or FX2 lanes after an ARP has been setup, the behaviour on whether the following FX interrupts the ARP or not is inconsistent. Sometimes, I've found that the ARP effect will stop, and execute the new command... but other times the ARP ignores the new command, and carries on. This happens both in FX1 and FX2, with any new FX command, and I haven't found a consistent way to get the same result. 

Expected behaviour: The effect in the relevant lane should be interrupted by the new FX command. 

### Bug Two: All three lanes required to change ARP FX parameters

If you want to change the chord that an arp is running, you can't do so without retriggering all three lanes (note, FX1 and FX2). If you try to change the Chord on its own, it kills the arp entirely. 

Expected behaviour: The other lanes should run as normal until interrupted by another command. You should be change the Chord in the relevant FX lane without it stopping the arp from running.

### Improvement/Expected Behaviour: No way to modulate arps.

Because of the way the ARP FX is implemented, you can't modulate the arp in any way. e.g. to modify the panning of an arp. This means that they are very static, and other tracker software provides ways to do this. 

Thanks

thebradclay, 2021-12-20:
Has this addressed?

MortalJS, 2022-03-19:
No response. Closed.

closed 2022-03-19 by MortalJS