polyend/TrackerBetaTesting #1072: [1.3.1] Buggy behaviour GridFX

static-noise-PT, 2021-05-26:
There is a weird issue when one is putting in FX values via the pads. It's jumping between values and sometime some values are skipped. It's hard to explain without video :D

On Discord the user dvdmasr posted videos and explained the issue as well.
The post is in #firmware-updates an is from  28.04.2021.

bndofc, 2021-05-27:
hey! i'm dvdmasr on discord, thanks for posting on gh, was too lazy to report it hahaha

here's the vid of the bug in action
FX value selection on the pads starts lagging behind, missing and skipping values. only happens in FX lanes
happening on 1.3.1 and 1.4.0b1

https://user-images.githubusercontent.com/80283211/119908019-a0f16c80-bf1f-11eb-93ed-520cf1f15f7e.mov

MortalJS, 2022-02-02:
Hi @static-noise-PT & @bndofc 

Can you confirm if this case still occurs in version 1.5.0

I am trying to reproduce the described case but with no result.

static-noise-PT, 2022-02-02:
> 

Hey. I hope I can find some spare time for music soon (next couple of days) I'll report back then.

spiralingalbatross, 2022-02-06:
This also happens to me.  On 1.5.0. 
New device and was wondering if it was hardware or firmware. Limited return window remaining, but must be firmware?

MortalJS, 2022-02-20:
Bug Confirmed.
@ziej8asek

bndofc, 2022-02-23:
i find it happens pretty consistently with the Roll FX

static-noise-PT, 2022-03-07:
Yep still there

closed 2023-04-21 by MiroPoly