polyend/TrackerBetaTesting #1074: [1.4.0b1] Performance mode pattern length issue

tfspsound, 2021-05-28:
I created a song where most of the patterns where 64 steps in length with the exception of 4 patterns that are 32 steps long. I hit play and went over to performance mode where the song played through fine until it reached the first 32 step long pattern which played through the 32 steps followed by another 32 steps of silence. It behaved the same way every time it reached a 32 step long pattern.

loktez, 2021-06-10:
When I engage a Pattern Length value in performance FX mode, regardless of the selected track (track 1 or track 4 or both for example), all other tracks are subject to the pattern length value (while they are not selected). It seems that this problem only appeared on this version 1.4.0.b1. My best

tfspsound, 2021-06-12:
Hey @loktez I think you have discovered a separate issue with pattern length fx. The issue I’m referring to happens when you have different pattern lengths regardless of what fx used in performance mode.

MortalJS, 2022-03-19:
No response. Closed

closed 2022-03-19 by MortalJS