polyend/TrackerBetaTesting #1075: more steps in patterns and ticks/tempo multiply

free4system, 2021-05-28:
Hey there guys, here's Beniamin :)

Your tracker feels damn good :) but there are two things which could provide more creative ways, especially in any jungle or just a breakbeat stuff.

When Im using octamed v4 or renoise on pc/mac I always go for 256 step length of patterns, as I see You provide us with only 128. If You could double it, it would be a killer as hell

second thing.. as I found out You can double the tempo in Your tracker, but in renoise and octamed, besides choosing tempo You could set ticks. it gave so many ways to create patterns differently especially with hardcore jungle and complex break patterns.

Hope those two things are reachable and doable among Your updates and fixes, it would really add the cherry on top of Your tracker

trutypesounds, 2021-06-04:
+ 1 yes please

antf4rm, 2021-09-08:
yes!