polyend/TrackerBetaTesting #1080: Saving instruments preset

CarloAvanti, 2021-06-03:
It would be a great feature to save instruments preset, to share between different project or even between different people. We could work on sound libraries specific for the tracker.
Or at least would be great to have the possibility to copy/paste Instruments parameters from/to different instruments.

anemlife, 2021-06-04:
You can already copy instruments + their parameters in the sample loader (use the copy/paste buttons). 
What you can't do is loading an instrument from another project

CarloAvanti, 2021-06-07:
Yes, and this is very useful, but in this case would be better to have the possibility to copy just one or more instruments parameter without coping the sample, for example, pressing REC in Instruments Parameters could let us select one or more parameters and copy/paste them to another instruments. That would be fun !

anemlife, 2021-06-07:
Got it, this could bring some new workflows yeah :)

bambrose, 2021-07-08:
Yes I realised this several times as a request. Single most obvious omission currently I think

MattWilsh, 2021-07-23:
+1

This is something I'd really like, and know a couple of other owner who would love this.

bambrose, 2021-07-23:
Single most important outstanding feature I think

Jacknametrouble, 2021-12-18:
This has been added in the newest 1.5b1 beta.

closed 2021-12-18 by Jacknametrouble