polyend/TrackerBetaTesting #1081: Another send effect for chorus

CarloAvanti, 2021-06-03:
It would be great to have some more send fx, something like a chorus (would be great to help the stereo image), a phaser or a flanger.

anemlife, 2021-06-04:
Definitely, a Phaser/Flanger would be really great! Especially for glitchy percs

The Sample Editor ones are not really intuitive as you need to find the "sweet" spots, then render, retry, etc... A "live" solution would be really welcome

k4ot, 2021-06-21:
yes please, live phaser and flanger
not rendered in sample. phaser written in the sample = phaser that never changes = useless phaser

"imagine dub music with same phase at each skank sample, it's very cool"  said no one ever

closed 2021-12-02 by Jacknametrouble

Jacknametrouble, 2021-12-02:
Thank you for your feedback, this won't be possible due to the limited resources.