polyend/TrackerBetaTesting #1082: Suggestion --> Line Per Beat OR bpm Multiplier

anemlife, 2021-06-04:
Most sequencers (eurorack, boxes, etc...) work around 250-350bpm. The tracker goes up to 800bpm.  

That means when syncing machines the tracker is somehow limited to other sequencer's speed. 

Trying to do fast melodies / grooves / drum fills / etc... is sometimes pretty hard on the PT compared to software Tracker's where you can set the LPB (to play for example 2 steps pers beat instead of 1). Which means you can type 300bpm drums on a 150bpm track with a software, but not on the PT. (really cool for jungle/breakcore/psytrance/even for drill, etc...)

I think there should be an option to play 2 steps per bpm on the PT so we could send a 150bpm midi clock and it would play at 300bpm (limited at 400bpm for example, to match the 800bpm tracker possibility)!

That's one of the biggest things I'm missing with the tracker.

ketumatek, 2021-06-04:
> Most sequencers (eurorack, boxes, etc...) work around 250-350bpm. The tracker goes up to 800bpm.
>
>
>
> That means when syncing machines the tracker is somehow limited to other sequencer's speed. [...]

I m used to play with Renoise and if I understand what you wrote in Renoise you can put the number of step as 8 16 32 and for the same tempo the tracker will read information as x2 or x 4 I think...
For example you are at 150 bpm on 8 step if you take 16 step you are always at 150 bpm but the tracker play as if it was at 300bpm 
Sorry if it not really clear (explane in English is not really easy)
But it should be a nice feacture if when we select in general option 4 8 16 or 32 step it will change the tempo time as divider...so 4 and 8 is normal tempo 16 and 32 equal x2 and x 4 
(Reference to Renoise)
Hope it make sens...

anemlife, 2021-06-04:
Yes makes sense! Would love some options like this (also values like x6 x 12 etc... for polyrythms would be fun but that's another point)

trutypesounds, 2021-06-04:
+ 1 for this option, a must have thing

tfspsound, 2021-06-12:
It would be nice if some of the fx like roll or the lfos could take into account bpm. Using a roll fx in a song with a high bpm currently is useless unless you want audio rate rolls. It sort of makes being able to write at high tempo redundant.

magnus-eriksson, 2021-12-25:
I'm new to this tracker (got it 2 days ago and really love it) and am already missing this feature. To get the beat I wanted, I had to put it to 266 BPM which cause issues with the TD-3 I had connected and synced through MIDI out. Remember back when I was using FastTracker 2, you had a BPM setting and also a Speed setting. The default speed was 6 (don't know what it was based on though) and I often put it on 3 to get double lines per beat to be able to do drum buildups etc.

Feels like it should be a fairly trivial addition so I hope you'll consider it (and give us some feedback here :-))

trutypesounds, 2021-12-25:
@wjakobczyk-polyend @Jacknametrouble @ambv @piotr3500