polyend/TrackerBetaTesting #1083: Pattern starts with 00 option like in classic trackers

trutypesounds, 2021-06-04:
Coming from classic trackers i was a bit confused that patterns starts with 01 mark, because FT2 or Renoise use 00 position  (see screenshot). Confusion comes from pattern divider where you can select 3/4/8/ etc, but when i look at the pattern itself the brain is so attached to even numbers like 00 / 04 / 08 / 16 ... its easier to navigate and use the pattern when u know that even numbers are where the kick is or sample usuallyy starts etc.

**00 ---**
01  ---
02 ---
03 ---
**04 ---**
05 ---
06 ---
07 ---
**08 ---**

But in polyend tracker its a bit confusing that it starts with 01, setting divider to 4 i expect to see my kick/whatever on even positions and not like 07/12 etc.

[image]

ketumatek, 2021-06-05:
> Coming from classic trackers i was a bit confused that patterns starts with 01 mark, because FT2 or Renoise use 00 position  (see screenshot). Confusion comes from pattern divider where you can [...]

Your question has already ask in an old post...
I m complete ok with you, I come from Renoise and it is a little bit disturbing and it will be very cool if they put this stuff as option in general content.
Choose between polyend tracker mode start at 01 
and Renoise mode start at 00.

AmiDonk, 2021-06-06:
Every tracker i have used start with **00** bcoz it make sense, polyend tracker does not, it make no sense.

bambrose, 2021-06-06:
not concerned me before.. but does makes sense to be honest.
would make navigating a 128 line sequence a lot easier.

trutypesounds, 2021-06-06:
I just visited the digitakt page on elektron's website and it clicked me - yes, polyend used this "pattern starts from 01" to attract groovebox users probably, 16 steps and all that stuff. So, @ambv would be cool to have this as an option in 1.4 firmware if possible.

ambv, 2021-06-06:
The option is available when hexadecimal numbers are used. We're not planning further configurability here as the amount of display options impacts realtime performance.

trutypesounds, 2021-06-06:
> The option is available when hexadecimal numbers are used. We're not planning further configurability here as the amount of display options impacts realtime performance.

thats sad news really, been tracking since 90s, but was never able to get used to hex mode, numbers easier to navigate =/

trutypesounds, 2021-06-06:
@ambv may be it will make sense to make "pattern starts from 01" in horizontal view mode, so its more groovebox oriented thing, and when in vertical "pattern starts from 00" ?

khidr9, 2021-06-06:
I think the way it’s set up is sensible as subdivisions of music.  “1” is the first subdivision that can make a sound after the barline no matter what meter we’re in.   It’s not what other trackers do and they’ve made it available in hex.    If it didn’t impact performance to iterate the numbers back to 00 I’m 100 for the ability to let users choose.  But if it would impact performance it should stay the way it is.  The observation that it’s more familiar to Groovebox users and anyone using western music theory for that matter is true and it shouldn’t be changed in a way that then removes that familiarity in the service of other users.   I’m saying this strictly on the idea that it’s a destructive choice due to performance costs.

ketumatek, 2021-06-12:
Related tread #866 and #375

It is regularly ask...

trutypesounds, 2022-01-16:
@Jacknametrouble merci

Mike-w78, 2022-02-21:
Could you reconsider this please?  If the option to switch the numbers to hex is supposed to be a 'throwback' to old trackers, IMO it fails in that aim because no old trackers used hex for the line numbers.  For someone who has come from protracker, octamed and renoise it feels alien and counter intuitive to have either 01 as the first line number, or to have the line numbers in hex.  I know a few other users who feel this way.

ambv, 2022-02-21:
The fact that ProTracker doesn't use hex numbers in row numbering is something I totally forgot. After I swapped the Amiga for the PC I spent most of my time in Fasttracker where rows were numbered in hex:
[image]

and in Renoise this is configurable:
[image]

But since the mighty ProTracker did it differently, well... I'll talk to the team.

Hailon00, 2022-02-21:
I am all for this option...

Lets not forget the might OctaMed also started with 00 and did not use hex.

Trying to retrain my brain after years of using 4, 8, 12, 16, 32 etc is a real mid melt. And in music those numbers are logical.