polyend/TrackerBetaTesting #1084: [1.3.1] A minor scale not correct

vanark, 2021-06-07:
Selecting the fill mode, and chosing the ‘a minor’ scale, the notes that gets filled are the wrong scale. ‘A minor’ has no sharps in it.

[image]
[image]

CarloAvanti, 2021-06-07:
Hi, this is not an isssue, you are not in A minor, you are still in C minor, just selecting notes between A2 and A3. To work in A minor you have to change the root note in project settings ;)

vanark, 2021-06-07:
Thanks. But it is an issue, as the general config just mentions “Pads root note”, not root note for everything. Also, that root note should be able to be set in the fill display, as it apparently is connected to the scales you are filling.

closed 2022-01-13 by MortalJS