polyend/TrackerBetaTesting #1085: More LFO/Envelopes

CarloAvanti, 2021-06-07:
I already posted soething about pitch envelope/lfo, but I was thinking in general the LFO/ENVELOPES section could be implemented.
It would be great to have instruments automation for tune, resonance, granular lenght, or maybe some assignable ones? maybe going in a "matrix" direction, giving the possibilities to modulate different things, maybe even using other instruments as sources or destinations... ??!!

;) :)

Marc-3000, 2021-12-30:
Pitch envelope would come in handy for making your own Kick and percussion sounds (from a single cycle wave).
LFO's and envelopes are always welcome ;)

shawnroos, 2022-01-14:
Would love to see a proper pitch envelope.

77377, 2022-09-13:
Agreed. More LFO shapes, more destinations pls.

CarloAvanti, 2022-12-06:
If the hardware is a limit, it could work also that you can select and activate only some of the LFO, now you have 6 (Volume, Panning, Cutoff, Wavetable, Granular, Finetune), you could have more to choose (tune (pitch), resonance, granular lenght, sends...) but only 6 tselected active in the instrument parameters... Also having more shapes would be very nice.