polyend/TrackerBetaTesting #1086: Sequencer for Perform Mode

CarloAvanti, 2021-06-07:
I know "perform" it' s something for live, but for me it would be great to have a sequencer to record perform automations in the song mode. In general more integration between perform and song modes would be great. I have also another idea but I will put it in a different post...

tfspsound, 2021-06-12:
#371 second paragraph is a related suggestion for introducing a preset system where combinations of fx could be recalled instantly. Either of these two suggestions would be very welcome

depaffect, 2022-02-14:
I just want to add my voice to this. If you could write FX parameters (assuming the FX lanes were free) in a recording perform mode, that would be amazing.