polyend/TrackerBetaTesting #1087: Indivudual partterns for every slot in song mode.

CarloAvanti, 2021-06-07:
Still thinking about integration between song and perform mode, would be very useful to select individual patterns in song mode, so every block could have a number on it, so that in slot 1 we could have pattern 1 for kick and pattern 2 for snare, 8 for hh and so on. This would give way more flexibility in song mode, but also everything would be more clear and at the end we will have less patterns to use.

k4ot, 2021-06-23:
very good idea, it would work like deflemask. less redundant data

zandergrin, 2021-07-02:
Also how closer to how renoise works, in a way. Love this.

Monsieur36, 2022-07-09:
I was looking in the manual and at videos to see if this was possible - the Song mode matrix seems to scream out for this. And perhaps other edits per cell e.g., number of repeats for a Track Pattern

TheGrizard, 2022-09-09:
I was just about to post something similar for this - would make building tracks/songs  a lot quicker as no need to copy and paste / create new patterns for every song event. I tend to use the tracker with each track for a song part eg track 1 kick, 2 snare etc so being able to choose the pattern per track and essentially mute tracks per song event and mix sequences from different patterns together would be amazing and save so much time!