polyend/TrackerBetaTesting #1090: [1.4.0b1] Corrupted Sample when powered via USB by MacBook 2015

pete223, 2021-06-07:
### 1. Steps to reproduce
- Plug Tracker power cable into Macbook Pro 2015
- Power on Tracker
- Observe first sample is corrupted

### 2. Steps to reproduce
- Plug Tracker power cable into USB hub
- Power on Tracker
- Observe first sample is not corrupted
- Plug USB hub into laptop
- Observe first sample has been corrupted

### Attachments and links
[image]

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
Often

* Does this bug happen in previous firmware versions?
Yes

* Does this bug happen in every project?
Yes

pete223, 2021-06-07:
Similar to https://github.com/polyend/TrackerIssues/issues/599 but with consistent steps to reproduce

anemlife, 2021-06-14:
Just to mention I had similar issues with a poor quality hub (full plastic, etc...) at a friend's place. Every sample from the session we recorded were corrupted because of his hub. Never happened with the one I use at home (a simple 4xUSB 3.0 but more solid built). You should maybe try with another one?

pete223, 2021-06-14:
I connected the HUB to a different non-apple laptop and had no problem so it's purely an issue with the Macbook.
Also, connecting to a newer MacBook Air, didn't present any issues.

I wonder whether it's my Mac in particular or if it's a common issue across all Macbooks of this model.

anemlife, 2021-06-14:
I personally use a Macbook Air 13' early 2015 and no issue with my hub! Strange...

pete223, 2021-06-14:
Ok cool, I've updated the title with the model as you're not having an issue with your Air.

Also I took a look at [this](https://www.pixelscope.de/videos/watch/6a014b7a-0006-41af-ba32-4ab1fe7cf01d) video from the comments of the linked bug #599 and he definitely plugs it into a macbook as well (could be a 2012/13 macbook pro?)

anemlife, 2021-06-14:
Last 3 things I can think of are:

- Did you tried plugging the tracker to the usb port that is the closest to the cable on your hub? 
(in my case for example when I plugged 4 usb powered devices into my hub, my AKAI APC40 started acting buggy. Plugging it to the closest usb port made it work fine again. I believe there's some sort of power loss when plugging too much devices that require power on such hubs)

- Did you try the other USB port to plug the hub? As far as I remember these macbooks got at least 2 usb ports

- Are you using the provided usb-c cable? Had issues with some other cables (not all of them though)

Hope you can find a solution!

closed 2022-02-01 by MortalJS