polyend/TrackerBetaTesting #1091: [1.3.1] Track and master volume issues

Munro13, 2021-06-11:
Hi,

I just got my Tracker and am having the following issues:

Tracks stop producing any sound for no reason whatsoever. The worst offender has delay fx and high pass fx. The only possible way to fix this is to turn the unit off and on, which is very tedious.

Mixer shows volume signal from tracks that have not even been touched / used at all.

The track / master volume ( not sure which) goes extremely loud for no reason, completely randomly. This is in pattern view and I’m not making any adjustments to volume. I have everything mixed well (fairly quietly to avoid any clipping) and then it just blasts out for no reason.

I have only used 4 tracks, stock samples and a couple of basic fx, the above issues are making it unusable for me...

bndofc, 2021-06-11:
not sure if you are using roll fx, but i can replicate the sudden extremely loud volume if you happen to quickly mute a track that's using RV (Roll Volume Up) commands. it happens completely at random. i will add this to #1029 . present on 1.4.0b1 also

Munro13, 2021-06-11:
Hi, no roll fx being used of any volume automation at all.

MortalJS, 2022-03-19:
No response. Closed.

closed 2022-03-19 by MortalJS