polyend/TrackerBetaTesting #1094: Ability to assign Midi Channels per Track (regardless of notes per step)[bug or a request]

sunfallz, 2021-06-12:
I use 1 of my trackers exclusively for midi sequencing external synths:
Currently, you can only assign a midi channel to an active step with a note inserted, and it's super fun to have a different midi channels per step on a track.. but, here's an example of this issue:
-create a 16 step pattern, with 4 notes on steps 1-4-8-12, assign those to midi channel 1, Exit edit mode 
(listen to your external synth :)
-return to edit track 1, and enter notes on empty steps of the same track, they end up with a different midi channel assigned to them automatically, thus not transmitting on the same midi channel as already set earlier.

(Or am I just doing something wrong, is this a bug or a request?)

Requesting that we can select an entire track or all 8 & Assign the Midi Channel for the track whether there are notes inserted or not, so that any note inserted will transmit on assigned midi channel when entering and exiting edit mode.

modulationsource, 2021-06-13:
I am not sure if I understand right.
You can choose an instrument per note in the instrument column. 
(Instrument M01-M16 at the end of the sample/instrument list)
So always when you set a new note it will be set with the actual chosen instrument,
but you can change the instrument in the instrument column.

sunfallz, 2021-06-13:
i see what you’re saying . 
what i just realized is the bug or request is different.
i’ll edit

craighitchings, 2021-06-15:
I would also like to be able to lock instruments to tracks (and only those tracks) for improved polyphonic live playing. Say I only want to use three tracks for an instrument to lock it to being three voices –and for accidental note holds to not spill over to other neighbouring tracks but instead steal a voice on one of those locked tracks. In the current implementation of spreading extra notes over to adjacent tracks, it’s very easy to create a complete mess.

craighitchings, 2021-06-16:
Additionally locking an instrument/midi channel to one track would be great for monophonic basslines. Possibly with last note priority so new notes replace older ones rather than spreading out across other tracks.

acidf2018, 2021-06-16:
https://github.com/polyend/TrackerIssues/issues/1031

ROM72, 2021-06-16:
+1. It would be very helpful indeed

clickysteve, 2021-06-17:
+1 to this.

sunfallz, 2021-06-20:
Can we get another 👍 +1 😃

sunfallz, 2021-06-25:
bumpsies for this critical enhancement 😄

sunfallz, 2021-07-04:
enhance.... 💯

Just-Peter, 2021-07-04:
+1 that would be a big midi improvement

sunfallz, 2021-07-26:
would love to see this in the beta! who's with me ? we've got 8 participants thus far...

sunfallz, 2021-08-09:
Can we get another 👍 +1 😃?!

closed 2021-12-21 by sunfallz