polyend/TrackerBetaTesting #1095: Suggestion --> Projects subfolders + renaming

anemlife, 2021-06-14:
Hello,

I think the tracker is missing a feature to put projects in subfolders. I like to keep organized and it's pretty difficult at the moment by having every projects in the same folder. 
Would love to have subfolders so we could do for example: bundled projects / artists projects / march projects / april projects / etc... etc...

Also being able to rename projects would be cool. Currently you can "save as" then change the name, then delete the older version but it's a lot of manipulations for something that could simply be available as an option on screen.

Thanks

Tuiman, 2021-06-15:
Yeah, the ability to rename would be very useful.

Midisplitter, 2021-09-20:
Fully agree on this

I just did my first live performance with the Tracker and everything went smoothly except for loading new projects. Since I can't keep the files I need for my live performance in one folder, I have to go look in my entire folder of files for the correct one while I'm trying to perform and this can be pretty stressful. I numbered each project I wanted to use for my performance and even thought that helped a little, it still wasn't ideal since the order of the projects seem to be random. So just being able to keep the files together in separate folders would be an immense help!

Thanks in advance

soy1bonus, 2022-03-01:
Yes, please! Just bought the tracker and this was a bummer!

e10k, 2022-04-25:
Is this on the roadmap by any chance? It looks like the projects browser was only tested with a small number of projects.