polyend/TrackerBetaTesting #1103: [Firmware v1.4.0 beta 2 and beta 1 and 1.3.0 official] MIDI sync out clock double tempo.

mashipx, 2021-06-18:
firstly appologies if i have approached this bug report in the wrong manner, im new to this.

basicly, my tracker is sending out midi sync in double tempo time, specifically to my RD-8 and TR8s drum machine and also my novation AFX.
All start/stop/resume fine but play at double tempo. all my usb and midi cable devices are coneected via a couple of MioXM modules. I must stress I have tested the tracker directly connected to each piece of hardware with no midi return to avoid looping issues and the problem persists. ive gone through all my hardware settings on all devices to find any problems and i have used other midi sync send capable devices directly connected to hardware and they preform as expected. so i believe i have ruled out any other device that maybe causing the double tempo issue. it only seems to be caused by my tracker. ive tried various routing configs in the midi config menu and nothing changes. is there anyone else having similar problems? i was hoping to use my tracker as the centre point for my dawless setup. but it looks like im going to have to invest in a hardware midi clock device istead, not the most ideal scenario. any help welcome guys.

thanks

breskar, 2021-06-21:
Have you tested clocking the drum machines from the Tracker midi out directly and not using the MioXM ?

mashipx, 2021-06-21:
Hi thanks for getting back to me... Yes i tried direct connect. Tho i tried again directly and it worked fine... I think i have rounded it down to my NDLR noodler box.. i had it on but not in use. It turns out its been sending out random messages and clocks even tho disabled.. it was even causing my other synths to randomly squeel at me and provram change. totally bizarre. To fix it all ive had to route all via my daw and mioxm units. The randomness has stopped but bang goes my dawless setup until i can get the tracker , ndlr and mioxm's to play nice..

closed 2022-01-13 by MortalJS