polyend/TrackerBetaTesting #1104: [Firmware 1.4.0b2] System Hang On Sample Import

bradybothwell, 2021-06-18:
Hiya!

I've encountered a bug with the new beta firmware, the system seems to hang and buzz when I import a sample while playing back a pattern. It seems to happen intermittently, and only once I've been going for a while.

Observed Conditions:
- had been playing back for a few minutes at least, doesn't seem to happen right away only after a while
- attempt to import a new sample while playback enabled
- enter a note
- This is when it hangs, power button non-responsive, needs hard reboot

How often:
 - Intermittent, Seems to happen once a session has been running for a while

bndofc, 2021-06-20:
had this happen today!

k4ot, 2021-06-21:
had this happen yesterday on 1.3.1

playback was running but the tracker display was frost, hardware keys were frost, power button was frost.
only workaround is to remove the power cord

bradybothwell, 2021-06-21:
@k4ot Were you using any  ~10 sec or greater length samples? 

 I'm trying to think of other habits or details that might create the condition:
- In both of the cases where it happened for me, I had multiple instances of an instrument with the same long sample in it,  close to the sample limit in length
- I copied/pasted the instrument
- I did not rename it in the new instrument slot
- The size of the file being imported when the crash occurs didn't seem to matter
- The crash happens when I returned to the pattern screen, it is possible it could be triggered by the first note firing on the instrument track

k4ot, 2021-06-21:
what i was doing :
-load a sample (6,7 secs) over an existing sample in the samplelist while playback was running
-press import
-crash happens while screen says "overwrite instrument?"
-tracker display frost, not any key doing anything, playback was still running