polyend/TrackerBetaTesting #1106: [1.4.0b2] Swing FX, Synced delay time not working when slaved to USB

bndofc, 2021-06-19:
I wasn't able to confirm if this is the case on b1, but the swing FX and synced delay time do not work when slaved via USB on 1.4.0b2. It ignores the assigned swing value, but If you are changing swing values just as the transport hits the swing FX, tracker will become completely out of sync with your device (also, tracker's internal tracks seem to go all wonky too, nothing like the normal swing FX behavior). synced delay time is also ignored, but i've had a few occasions where it does work.

MortalJS, 2022-02-22:
Hi @bndofc 
Can you check if the problem occurs in firmware 1.5?

bndofc, 2022-02-22:
hello @MortalJS, i can confirm swing FX is now working as expected when synced via USB in 1.5.0, but synced delay is still broken !

also, could i go back to issue #1117 and report my findings? i've got the time to test stuff now!

natedey, 2024-09-12:
Are y'all sure this is working correctly? 
I haven't been able to get swing FX to work, and I purchased my mk1 tracker in mid-2023.
https://backstage.polyend.com/t/swing-with-external-usb-clock/5006