polyend/TrackerBetaTesting #1107: [1.4.0b2] Volume FX value used for controlling MIDI velocity carries over to an internal instrument after instant pattern switch

bndofc, 2021-06-19:
I was sequencing some of of my MIDI gear on Track 7 and using the volume FX to control its velocity. on the next pattern and on the same track, i had sequenced some internal sounds from the tracker only, with no FX applied. after changing patterns (via the instant pattern switching, not cued up) some of the Volume FX i had on the 1st pattern carried over to some of the steps of the next pattern, in my case absolutely blasting my ears as i had some Volume FX at 100. can't really seem to replicate it consistently, it just happens at random.

MortalJS, 2022-02-02:
Hi @bndofc 

I am trying to reproduce the described case but with no result.
Can you attach a simplified project illustrating the situation?

MortalJS, 2022-02-20:
Closed. No response.

closed 2022-02-20 by MortalJS