polyend/TrackerBetaTesting #1111: [1.4.0b2] Step Repeater fx in perfor

trutypesounds, 2021-06-20:
## Hi, sorry you had trouble!

### Steps to reproduce

Please describe the steps to reproduce the bug. Tell us what you noticed
happening and what you expected to happen instead.

### Attachments and links

Often the best way to show the problem is to record a short video,
attach the Tracker project you were working on, and provide links.

Pro-tip: You can drag and drop images and `.zip` files directly to the edit
box when writing your bug reports. Videos are too big as attachments, you
will have to use an external service for this and provide us links. Please
don't use services that delete files after short periods of time.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")

* Does this bug happen in previous firmware versions?

* Does this bug happen in every project?

tfspsound, 2021-06-23:
I experienced this too but it was a while ago, and currently it seems to be ok. I’m running 1.4.1b and it doesn’t seem to happen on that, so far.

trutypesounds, 2021-07-02:
where did u get 1.4.1b ???

> 
> 
> I experienced this too but it was a while ago, and currently it seems to be ok. I’m running 1.4.1b and it doesn’t seem to happen on that, so far.

trutypesounds, 2021-07-02:
now this is even weirder, check the attached video - there is nothing on channels 5678 at all, but when i mute them repeater works as intended, when unmuted they produce that buzz

https://user-images.githubusercontent.com/17484006/124283381-84ff6d00-db54-11eb-9736-60cfa451aae8.MOV

here is the project itself
[techstep098.zip](https://github.com/polyend/TrackerIssues/files/6754960/techstep098.zip)

trutypesounds, 2021-07-26:
Well, this is really annoying now, exactly step repeater messes something up here! 
As you see in the video, the first 3 channels are playing chord (wavetable instrument), 456 are TR909 drums, 78 is empty.
In performance mode channels 45678 activated, if i start playing with step repeater, the instrument which is playing the chords (on channels 123) somehow leaks into channels 7-8 and starts to play continuously thru the song.

Clearly this is a bug, as in previous videos shown above, some of the instruments starts to "leak" into empty channels when ster repeater activated in performance mode. @ambv please confirm this as a bug

https://user-images.githubusercontent.com/17484006/126994915-3bb7574c-1a84-4977-ad3a-bda3a9f0b8b3.MOV

also i've attached the project itself here
[atx-even.zip](https://github.com/polyend/TrackerIssues/files/6878463/atx-even.zip)

trutypesounds, 2021-09-06:
why this still not a confirmed bug? @ambv @Jacknametrouble

trutypesounds, 2021-12-16:
well, it's still here in 1.4.1b1 guys @ambv @Jacknametrouble @ziej8asek @wjakobczyk-polyend please pay some attention to this!

https://user-images.githubusercontent.com/17484006/146426037-c4273af6-43d0-4517-9274-cd879d10a750.MOV

trutypesounds, 2021-12-25:
@piotr3500 pls

Jacknametrouble, 2022-01-15:
@MortalJS any luck with testing this or no luck with reproducing it?

MortalJS, 2022-01-19:
Duplicate with #1364 > Closed

closed 2022-01-19 by MortalJS