polyend/TrackerBetaTesting #1113: discard feature in sample editor

k4ot, 2021-06-21:
i'd like to discard some parts of a sample.

i can crop a sample but i cant discard/trim some parts of it.
i cant find it anywhere. 

exemple:
here's my sample 1-2-3-4-5-6-7-8-9-10-11-12
i'd like to trim parts  5, 6, 7, 8, 9, 10, 11
so my final edited sample be like 1-2-3-4-12

right now i need to edit the samples on my pc ?
this makes the sample editor basically totally useless for me, its lacking the most basic feature ever

tfspsound, 2021-06-23:
It would be nice to at least be able to “mute” clips from a Slice or Beat Slice so that only “unmuted” clips where available across the pads.

k4ot, 2021-06-23:
yes so unneeded slice dont get written in the track with "fill" function.

or.....maybe code-wise its just simpler to add the ability to cut inside the sample to get rid of unneeded slices, and that would also kill less ram.
i say it again, dont want to sound rude, but this is a very very basic feature of every sample editor. even amigas had it back in the 90s.

darenager, 2021-07-03:
@k4ot did you try to create a pattern with just the slices you want to keep, then select them in the pattern then render? Saves needing to use the computer to edit, it is a workaround but will do what you want.

k4ot, 2021-07-03:
Thanks for suggestion but imo sample cropping should take 3 steps. Choose start, end, and discard! . Done. Something that basic should need no workaround, Yours works but takes xxxxx steps. It's like... Using a whole truck just to deliver one single bag of crackers.
This was possible on 90s computers.

k4ot, 2021-07-04:
OK for a workaround but it still takes far too much steps for me (and I don't even talk about resampling that might affect the sound).
This should take 3 steps : select start, select end, do!
Anything taking more than 3 steps is overkill. Again this is a very fundamental feature that was available in a computer released in 1985, and also in every sample editor