polyend/TrackerBetaTesting #1114: change the step repeater behaviour

k4ot, 2021-06-21:
FIRMWARE : 1.3.1

right now step repeater has a bizarre behaviour
it should repeat the X steps playing before activation, whatever the track is playing, and never change.
the repetition should stay the same, like MPC one step repeater feature, or like in Phutboyslim VST by Scuzzphut.

this is not the case. what is in the track has an influence on the activated repeater and should not, as long as the repetition is going on.

tfspsound, 2021-06-23:
The step repeater is very predictable imo it’s working in real time so as steps are read they are repeated until another sample is triggered or it receives an OFF command or you deactivate the fx.  Hope this helps

k4ot, 2021-06-23:
maybe i got the habit from the MPC repeater too much.
when you activate it, it just repeats the same thing forever until switched off, and doesnt care what is written in the track. the behaviour is very different on PT