polyend/TrackerBetaTesting #1116: Beat Slice: auto-slice into even slices

piotrMocz, 2021-06-22:
### My workflow problem and workaround

When slicing breakbeats (as a lot of Tracker users do) I often find myself in a situation where I want to slice the beat into even parts (8 is my go-to number but I suppose any power of 2 is useful). Since the `Auto Slice` option seems to slice on transients, it often produces very uneven slices for more syncopated beats. That is very useful in some cases and quite unfortunate in others.

As such, I find myself using the following workaround (which admittedly may be quite weird): I check the sample length and then run a simple Python script:
```python
for i in range(1, numberOfSlices):
    print(sampleLength / numberOfSlices * i)
```
This produces a list of breakpoints that I later manually enter in the slicer 😂  As stupid as this may seem, it goes a long way towards showing that it's quite easy to implement.

### Proposed solution

Add a menu under the `Auto Slice` button that will include the `Even slices` option and the ability to select anywhere between 2 and 64 slices. The menu itself could reuse the one that's used with the `Fill` option in the pattern editor.

P.S. I think I could actually implement that, let me know if you need any help. This would make my workflow with the Tracker a few times faster, so I don't mind spending some time coding 😄

bambrose, 2021-06-22:
It's already there . Just hold shift when selecting slice (it's in the manual)

piotrMocz, 2021-06-22:
My bad, closing.
Thanks @bambrose

closed 2021-06-22 by piotrMocz