polyend/TrackerBetaTesting #1119: [1.4.0b2] Line Delay and Line Reverb only work with high input volume

eingedanke, 2021-06-22:
Line Delay and Line Reverb only work if the volume of the sound coming from Line In is somewhat high.
There seems to be a volume threshold somewhere, below that Delay and Reverb don't kick in.

I could imagine that you guys have a gate built in to switch the DSPs off to save processing power when there's no (or a low volume) input signal, but this threshold is way to high, somewhere around 50% of the Line In VU meter!

talktogreg, 2021-06-23:
Was only minutes away from making the same report, so can CONFIRM this issue.

PITmast, 2021-06-23:
Hi

Additionally, I've noted the only one channel it's sent to effects when stereo is selected in configuration. But it's not such a big disadvantage I think.

Regards

joeyfivecents, 2021-07-02:
Confirming this behavior in 1.4.0b2

trutypesounds, 2021-07-02:
was mentioned here #1109

eingedanke, 2021-07-05:
> was mentioned here #1109

Yes, although it's not because of decay time as mentioned in #1109 but because of volume threshold IMHO. Devs should be able to confirm this.

trutypesounds, 2021-07-05:
> > was mentioned here #1109
> 
> Yes, although it's not because of decay time as mentioned in #1109 but because of volume threshold IMHO. Devs should be able to confirm this.

yeah, agree on threshold thing

MortalJS, 2022-01-13:
#1211 Duplicated

closed 2022-01-13 by MortalJS

joeyfivecents, 2022-01-13:
Hello. Is there any way you guys can send me the new beta firmware? Apple
lost all of my keychain info for a lot of sites including Github so I can't
log in and don't want to start a new account and Github is acting strange
so I can't see any of the info. Thanks.