polyend/TrackerBetaTesting #1120: Main Out/Headphone Limit for Hearing Protection

craighitchings, 2021-06-23:
Please add a Config option to set the volume of the Main Out/Headphones as a percentage of the master volume to protect our hearing. For example, I would like to set a limit of 50% which cannot be exceeded. This is to prevent hearing loss from unexpectedly loud sounds that occasionally happen when using Tracker, such as when Previewing samples with the volume too high, for example.

alexandernoakes, 2021-06-24:
Fully needed, so many times have had hearing shock when previewing as when sub mixing tracks you tend to allow for headroom and then going back to the sample loader it's then too loud.  a Preview -12db or something?

doelman, 2021-06-24:
Indeed, or a seperate volume control for the preview option in the sample
loader would be nice as well. It's so bloody loud and lowering the volume
is disturbing the workflow a bit.

AmiDonk, 2021-07-03:
This is professional music equipment and does not have sound attenuation. It's everywhere to adjust. You can not put such on mixers or amplifiers. They do it on video games, mp3 players and other things, but not on professional music equipment.

craighitchings, 2021-07-03:
That’s not true. It’s on Elektron’s Model:Samples and Model:Cycles, for example. And seeing as instrument volumes all need to be reduced because of the ease with which it’s possible to clip the master out, sample previewing is always far to loud compared to other instruments while working on a track.

eingedanke, 2021-07-05:
Same here, I tend to lower the volume of all instruments to around -12db to not clip the master bus.
A gain control for sample preview is urgently needed, not only to protect our ears but also to preview samples with a reasonable volume and to make them blend in to the volume of the song.

trutypesounds, 2021-07-06:
absolutely agree on gain control for sample preview, its much louder in headphones, than on monitors, pisses me off alot

astrochili, 2023-03-22:
I agree. I think the master level in the project can quite rightly be zero, but we need a general headphone output level setting.