polyend/TrackerBetaTesting #1121: [1.3.1] Aliasing due to auto-conversion

ROM72, 2021-06-24:
I imported this on the tracker with file transfer mode : 

https://www.dropbox.com/s/jftjn060c3tzkn2/ALBIPIED_MARMOTTE.wav?dl=0
and when I played it on my tracker there was some nasty aliasing on it.

https://www.dropbox.com/s/sqwbcx1d8wlwb3t/Marmotte-from-tracker.wav?dl=0 

the original file is 48khz 16bit.
I tried to play it with anti-aliasing on and off, but it's really caused by a bad conversion I think.

wjakobczyk, 2021-06-24:
How did play the sound on tracker? In sample loader preview? If so, then this explains the issue, the preview uses only simple antialiasing. A more sophisticated algorithm is used during actual import of a sample (then conversion not need to be done in real-time).

ROM72, 2021-06-24:
The second link is a recording from a pattern, it's not a sample loader preview.

bambrose, 2021-07-03:
Can you repeat with a 16bit 44Khz version of the same sample ?
Is the problem limited to 48khz sample coversion ?

MortalJS, 2022-01-20:
Hi @ROM72 

Have you tried the suggestion of "bambarose"?

closed 2022-02-01 by MortalJS