polyend/TrackerBetaTesting #1123: [1.4.0b2] Roll Fx behavior

anemlife, 2021-06-26:
A few things with the roll Fx feels strange: 

1) It takes priority to some Fx, for example: if you apply a Band Pass Fx on top the roll takes priority so you can't do a "filtered roll". I think that's missing

2) RV really often results into inaudible rolls (it was working better in previous releases if I remember)

3) Rv feels inconsistent with external clock (using a Multiclock for a super tight clock)

4) Simple R16 rolls feels even more inconsistent with external clock => full of glitchs/tempo adjustments

Is it just me or could anyone check if the Roll Fx works better for them?

Trying to do these super edited and fast trap hihats but it's way too bugy :(

Thanks

bndofc, 2021-06-26:
can confirm this behavior too

roll FX behaves kinda funky too if there's swing in your pattern. it will also bug out if you use Roll + Micromove FX, it won't trigger whatsoever.

anemlife, 2021-06-27:
> can confirm this behavior too
> 
> roll FX behaves kinda funky too if there's swing in your pattern. it will also bug out if you use Roll + Micromove FX, it won't trigger whatsoever.

Thanks for checking! 
Hope they make it work better then, it's a massive workflow killer for me :(