polyend/TrackerBetaTesting #1125: [1.3.1] Tracker turned off still takes 80mA of current

x2db, 2021-06-27:
Can this be solved with FW update ? 
Tracker takes ~80mA when turned off. So I got to the habit to unplug the USB cable from batteries b/c in a few days the batteries are almost drained. Tracker must be my first USB powered device that needs that much power in the OFF state.
[image]

talktogreg, 2021-06-27:
Kinda agree. It doesn't indeed draw enough standby current to prevent most USB power banks from auto-powering off.

zkurtis, 2021-11-13:
Super annoyed this still happens. You should at least warn user's. I kept thinking my batteries were bad until I looked it up.

trutypesounds, 2021-11-19:
Its still in 1.4, drains power when off

laxmodin, 2022-02-21:
Yep!