polyend/TrackerBetaTesting #1129: What does Gate FX do?

prds100, 2021-06-28:
I dont know if this is a bug or feature request, just a question.

The Gate FX is kind of a mystery to me, what is it actually doing?  I assumed it was linked to sustain or decay in the envelope but it doesnt act like manually adjusting those parameters would do. On a sliced beat it triggers different lengths for each slice even though the gate value is the same across the track, sometimes a slice wont sound at all with gate active?

Anyone can shed light on this?  I love a good gate in my tracks so if i can understand what to do with it that would be great.

ketumatek, 2021-06-29:
There is another topic about this but I didn't find it...nether mind.
I m agree with you the gate work at 6-7 under there is no sound and upper the sound stay normal.
I m use to use gate on drum slice to cut the slice...and I think it should be more precise because as I said up to 8 it doesn't affect the sound and under 6 no more trig...
So as you say ...what the gate is for?
I think the team have to work on it for more accurency...

bambrose, 2021-07-03:
Reduce the release of your volume envelope to zero.
i.e. use decay and sustain portion of the envelope only and you will hear results like those you're expecting I think.
As @ketumatek says though the 'scale' or values are a bit off.

closed 2021-12-02 by Jacknametrouble