polyend/TrackerBetaTesting #1132: [1.4.0 Beta 3] keys unresponsive after boot

aheger, 2021-07-01:
### Steps to reproduce

Installed 1.4.0 Beta 3.  
Unit boots normally into Pattern screen. Keys are totally unresponsive.  
The data knob works. The pads light up. Other keys do nothing.  

Resetting back to Beta 2 fixes the problem.

PITmast, 2021-07-02:
Is it possible to go back to Beta 2 with keys issue?

aheger, 2021-07-02:
> Is it possible to go back to Beta 2 with keys issue?

I used the Reset-Button to get back to Beta2. See: https://github.com/polyend/TrackerIssues/blob/master/HOWTO_BETA_FIRMWARE.md

PITmast, 2021-07-02:
Anybody else can confirm keys issue?
I want to test Beta 3 but can’t use Polyend Tool in case of problems due to Mac OS version.

Regards
Piotr

k4ot, 2021-07-03:
Had a similar problem with 1.3.1 while I was in the sample loader. It bugged, then every square key was KO.
Light leads OK, jog OK, power button OK, but not one square plastic key was working anymore.
Somehow a reboot doesn't fix the issue.
Only fix is to unplug power cord and plug it again.

To me it looks like a software bug that became hardware bug. The fact the bug is reboot-proof is very strange. 
I asked on PT if someone had the problem. No one.

I RMA'ed the PT.

bndofc, 2021-07-05:
i recently had this happen on 1.4.0b2, don't know under what circumstances it occurred. indeed the only way was to fix it was by unplugging it

PITmast, 2021-07-10:
I have no keys issue with 1.4.0 Beta 3.
Looks like #1115 and #1098 are fixed.
Thank you Poly for this update.

Regards

closed 2021-11-08 by ziej8asek