polyend/TrackerBetaTesting #1134: [Firmware version - ALL] Select Track Contents selection is broken if sequencer is running

bambrose, 2021-07-02:
This bugs the fuck out of me sorry.
Has been flagged as 'resolved' before, but isn't.
Check out the video
`https://www.dropbox.com/s/q3v7vvzpb5114qp/TrackerIssue.mov?dl=0`

Steps to reproduce.
1. Record events on a track
2. press play on the tracker
3. hit record
4. select shift + insert (home)
5. select shift + up (to select contents of current track)
6. hit record
7. hit record
7. select shift + insert (home)
5. select shift + up (to select contents of current track) << it selects contents of all tracks.
6. This is a bug

Thanks

bndofc, 2021-07-06:
okay, great to know it wasn't me going crazy!! i was wondering why no one had brought this up

closed 2021-12-18 by ambv

bambrose, 2021-12-18:
Yes !! Thank you so much for resolving this