polyend/TrackerBetaTesting #1136: [1.40b3] Tempo jitter and midi clock jitter

darenager, 2021-07-03:
### Steps to reproduce

Internal tempo jitter:

Record a simple pattern with 4 to floor kick at 120bpm, record the audio output or export, load the wav file into an audio editor on your computer, set markers for each beat using tempo 120bpm - For this test I used wavosaur which is a free windows audio editor, I trimmed the file to be exactly 8 beats, added the markers using set markers at 120bpm, each hit should be 22050 samples apart (@44.1khz sample rate) 

What I found was that some beats were late, it is fairly reasonable to expect a slight deviation, however in some cases the beats were off the markers by upto 171 samples which is roughly 3.8ms, which is clearly audible. You can easily hear it even if you listen to the loop in isolation, but it gets much easier to notice if syncing to/from other gear. Which brings us to midi sync jitter.

Midi sync jitter:

Now set the tracker to sync to midi of a known stable midi clock, in this case I used a Roland SBX-1 sync box set to 120bpm, I used the same method outlined above, this time the result was a little better, I experimented with different clock delay settings but it did not significantly change the result, this time the beats fell slightly early with the worst being 90 samples off or roughly 2ms. I also tried syncing tracker to other drum machines with known stable tempo (because I tested those) and the result was that the beats would smear or flam when playing same beats as tracker.

As is well known consistent lag can fairly easily be accounted for by offsetting events or clock, but jitter can’t be fixed other than by firmware updates.

### Attachments and links

I am unable to post a screenshot or video at the present time, but if required I can do so later, in the meantime it is pretty easy to verify my findings following the steps to reproduce listed above.

### Other important information

* How often does this bug happen? (every time, often, "it only happened once")
* All the time

* Does this bug happen in previous firmware versions?
* AFAIK yes

* Does this bug happen in every project?
* Yes

bambrose, 2021-07-03:
That's pretty significant internal jitter. I didn't realise

bambrose, 2021-07-03:
Looking at render files for kick drums, it's clear to see that a straight four to the floor does indeed speed up and slow down.
Not quite the same issue (that's the computation of the render ?) but same net result

darenager, 2021-07-03:
@bambrose yes it is pretty significant, I don’t claim to have really sensitive hearing, but I can hear the fluctuation if I listen for it, more easily noticed on a sparser rhythm, and once you hear it it is difficult to unhear 😆 

Try this, just put in a 4 to floor kick beat nothing else, tempo 120bpm, then just listen for a little while, you can definitely hear when a kick is slightly off time. But obviously much more of an issue when trying to sync with other gear.

darenager, 2021-07-03:
@bambrose yes could be due to the 44117 export rate, however it is still not right if recording externally at 44100. I edited the original post to exclude exporting and just record externally to make it clearer.

sebpmtl, 2021-10-03:
I have the same problem. I even made a test syncing an LXR-02 with the Tracker both at 130.0 bpm, and starting them at the same time, it's tight for a while and the tracker starts lagging eventually. If I sync my Erica bassline to the LXR(that's synced to the tracker), I can see the bpm jumping from 130 to 129 constantly..

darenager, 2022-02-23:
@MortalJS yes it still occurs, additionally the clock sync delay is not able to correct it and at the setting of -7 it was ahead of the master but at -6 it was behind, and at -6.5 it moves forward continuously relative to the master. I hooked it up to my oscilloscope to determine this, it is definitely a bug. BTW the master clock was an Elektron Octatrack mk2 running the latest current firmware (1.40A) my scope was set to a 50ms window, the Octatrack has a very stable midi clock, verified by syncing with other gear, see the attached video, the Tracker is set to a delay of -6.5 so you can easily see the movement against the master. Direct midi din connection OT out to Tracker in, using the supplied TRS adapter.

https://user-images.githubusercontent.com/33150745/155403135-5d875523-b83e-4ff6-aea0-5fbc57f0391a.MOV

darenager, 2022-05-02:
Just tested with 1.6.0b3 and the midi sync is still not working correctly - I still can’t get it to sync properly with any devices when tracker is synced via midi.

JoeDobise, 2022-07-04:
I have been following this bug report since I am experiencing the exact same problem as @darenager, to validate I did not use a multimeter I have just used the audio output. As a sanity check I configured an octatrack mk2 as a follower from a stable midi clock initiated via bitwig, which for myself was around 1 ms variance of audio event onsets over ~200 audio events(I will consider this variance how I am measuring jitter, these are all approximations). 

Using the thru from the Octatrack mk2 I setup the Polyend Tracker as a follower from the midi clock(initially being produced from bitwig), multitrack recording the audio output of both devices with the same sample sequenced in the same manner, the results displayed the following type of events: 
- Octatrack behaved similarly to the baseline with jitter of ~1 ms over 200 events, the Polyend Tracker saw ~10ms over the same 200 events
- the pattern of the drift was identical to what @darenager recorded above, an evergrowing shift to the right

Additionally I tried all of the following configurations, all of which provided varying levels of jitter greater than or equal to ~10ms:
- swapping the midi cables for the inital setup of bitwig -> blokas midihub(over usb) -> Octatrack mk2-> Polyend Tracker 
- setting the Octatrack as the initiator of the midi clock with the Polyend Tracker as the follower
- setting up an Analog Keys as the midi clock initiator into midi thru configuration via a blokas midihub to Polyend Tracker and Octatrack 
- using the usb connection on the Polyend Tracker for clock with the source produces from bitwig

I tried to use the Polyend Tracker as the initiator for clock and it appeared to produce a similar level of jitter, with a similar shifting to the right pattern but I did not experiment too much there.

The results of this behavior is I do not use the Polyend Tracker with other devices when sync is important, and mostly only use the device as itself adding in previously prepared loops.

I am curious if anyone else has looked into this to validate the behavior.

darenager, 2022-07-07:
@JoeDobise I have not tested with Tracker as master device, I am unable to do any testing for a while, but when I get the chance I will test this also.

c43d35, 2022-12-28:
not sure if still necessary, but i'm also here to say clocking from external midi sources drifts away after time.. tracker is on fw 1.6.. tested with ableton via focusrite and also a vector sequencer.. it feels like it's getting worse (drifts faster away) when sending cc messages to the tracker while clocking.. but i've no numbers to prove this

JoeDobise, 2022-12-28:
@c43d35 The cc messages would make sense to me here but I have not done any testing with that. Additionally something else I have been thinking is I know Squarp recommends only using [specific microSD cards](https://squarp.net/rample/manual/#LLevGNv) with the Rample, if my memory recalls correctly this was due to the read rates being inconsistent across a large set of microSD cards they had tested. Not sure if this could be playing a factor here or not.

This is definitely still an issue for me, sucks cause everything else about the Tracker is just too damned good for me to let go of it. I pretty much only use it on its own since I cant trust the clock to interact with anything else that uses a clock (generally I see a drift/jitter of 10ms within 200 beats).

closed 2023-04-21 by MiroPoly