polyend/TrackerBetaTesting #1137: [All Firmwares] Decay Effect Feedback

bambrose, 2021-07-03:
For my money, with Delay feedback set to zero, there should still be some effect. a single repeat.
The delay feedback circuit is how much it feeds back to create a chain of echos, but the effect itself is a single delay of the original signal. However this is not how the delay effect works if I am not mistaken..

trutypesounds, 2021-07-05:
you talking about master fx or offline fx?

both delays are crap imo, no filter modulatons and no endless fx (for dub fx etc)

bambrose, 2021-07-05:
I'm talking about the master fx. I don't use the offline effects.. I find it hard to think how an offline delay would work unless it quadrupled the length of the sample or something

OmskInfo, 2021-07-16:
Yes, I 1000% agree (and suggest to rename this bug to **Delay** Effect rather than Decay effect, if possible).

With feedback at zero, I should just hear one instance of the delay. Only when I turn up the Feedback should I then start to hear additional repeats.

MortalJS, 2022-01-20:
Hi all,

I checked the described case and compared with different three external delays. I confirm that the delay in the tracker works a little differently. 

I will classify this as a bug. We will see what we can do about it.
As for the endless delay, at this point I can not guarantee anything.

closed 2023-04-26 by MiroPoly