polyend/TrackerBetaTesting #1144: Some ideas for workflow enhancement

darenager, 2021-07-07:
Here are some ideas that would enhance workflow, I hope they might be considered:

Rotate track - the ability to move all track events incrementally by step forwards or backwards with wrap around, for example in a 32 step pattern the whole track can be shifted such that what originally was on step 1 is now on step 28, step 2 is now 29 and so on. Handy for moving sequences to be “one the one” if created randomly, or for offsetting for different sing sections etc.

Loop section of pattern for editing, enter/exit loop mode by shift + pattern button, screen button 1 turns loop on/off, screen button 2 selects start step of span, screen button 3 selects loop span length in steps, buttons 4 & 5 move loop < >, button 6 sets loop move amount in steps, 7 & 8 double/half loop span. This would give every reasonable option to quickly edit sections of longer patterns I think. Could be useful in performance too.

Export track option, to allow a track from another project to be imported. For example sometimes I change a project, save as new, listen back to earlier version and maybe I liked the old track 1 better than the current one, so I go to old project, export track 1, go to current project and import it.

When exporting audio the option to set rate at 44100. Also in sample editor option to export at 44100 - great for making samples for other gear. When exporting pattern, option to name individual tracks as track name rather than t1-t8 defaults. Tracks without events not exported.

Consolidate after purge samples to move all instruments to concurrent slots, with pattern data updated to reflect the change.

Overview of instruments, shift + press instrument and all 48 are shown on screen in spreadsheet, soft buttons select level, pan, sends, etc to allow easy overview and get mix right. Use pads or arrows to select an instrument.

For “where” in fill mode when note or fx is selected have second column with option to select specific fx or instrument or note only. Also context sensitive NOT, so you want roll on every instrument except instrument 3.

Fill option to remove specified notes, with range.

When selecting multiple tracks the ability to exclude a track by pressing a soft button. Or muted tracks don’t get copied option?

Option to have easier to see pattern divider, at some viewing angles it can be difficult to see.

Auto preview on stepping manually, maybe have preview button latching? And always have chance ignored when previewing.

Config option to only have the selected track(s) armed when wanting to realtime record.

Realtime erase/delete.

Spot record - hold record and hit note at time desired to enter into playing sequence, so holding rec button allows the record head to continue to move. Also can be used for realtime erase/delete.

Have (option for) performance buttons momentary, and have off as off rather than wasting first row for off. A press could be momentary, shift + press to latch.

Notes for song, a text file which is saved with a project.

frognerboy, 2021-07-22:
This...

"Overview of instruments, shift + press instrument and all 48 are shown on screen in spreadsheet, soft buttons select level, pan, sends, etc to allow easy overview and get mix right. Use pads or arrows to select an instrument." 

...is genius!

juandale77, 2021-07-24:
You can /almost\ do the pattern rotation idea using the insert button, but I think what you've described here would be more useful.

Love the pattern looping idea! That would be great for performance.

Thanks for sharing your ideas!

darenager, 2021-07-25:
Yeah if insert/backspace ‘wrapped’ around pattern boundaries that would do the trick, maybe hold the arrow up/down key along with insert to do the function.

TwistedGenetics, 2021-07-31:
I would love to have the ability to loop part of the patten. Makes edits at the end of bars easier

bambrose, 2021-07-31:
This is a great idea.
Preview is a bit too clunky as you can't just leave it looping while you edit

SimmeSimon, 2021-09-12:
Rotate track with wrap around would be really great! If you could highlight a track or multiple steps/tracks and use the jogwheel or the arrow keys to rotate  inside the highlighted section (maybe not just the whole track/pattern but just in the highlighted part) could yield fast inspiring changes or preferred corrections. Absolutely was searching for this since day one :-)

bambrose, 2021-09-12:
Yeah agreed. Would be a very insuring addition

closed 2021-12-02 by Jacknametrouble