polyend/TrackerBetaTesting #1146: “Bake in” play mode

darenager, 2021-07-07:
The performance function “Pattern play mode” can lead to some really great variation in track data, it would be really great to have the ability to render new track data from this, especially the C1-C20 type.

So imagine you have a track(s) selected for performance mode, you find say C5 gives a nice variation, hold performance button and option pops up to copy current pattern to new slot with the C5 modifications turned into note data for the selected tracks.

I think this would be a really nice way to come up with a bunch of variations.

thumperjack, 2021-12-14:
that would be awesome

Jacknametrouble, 2022-01-16:
This won't be possible. Thank you for sharing your idea.

closed 2022-01-16 by Jacknametrouble