polyend/TrackerBetaTesting #1150: [1.4.b2] right side audio output distorted

joeyfivecents, 2021-07-09:
## Hi, sorry you had trouble!

### Steps to reproduce

If the Tracker is left on for at least 30 minutes, when I start a new project, the right side audio is distorted. I tried it with multiple headphones and audio interfaces. A reboot/restart fixed the problem.

### Other important information

* This just happened and has not happened before or I just didn't notice until now.

* Does this bug happen in previous firmware versions? No

* Does this bug happen in every project? Yes

Jacknametrouble, 2021-07-09:
Sounds like it may be more of a hardware issue (codec?). Make sure that isn't caused by volume levels too high and reach out to https://help.polyend.com/submit_ticket if necessary.

joeyfivecents, 2021-07-09:
I'm hoping it was a one time thing. If it happens again I will submit a ticket. The only thing that indicates it might be Tracker is the fact that I plugged in a bunch of different headphones. It might be nothing though. Gremlins maybe.