polyend/TrackerBetaTesting #1152: Change pitch without triggering sample in notes

darenager, 2021-07-10:
It would be nice to be able to add a pitch change in NOTE column without triggering the sample again - would be really handy used with glides for acid/basslines I think.

Maybe go to a note, hold the note and press shift to grey it out, this would indicate just a pitch change?

I know we can use one of the FX slots for tuning to get a similar result but this proposal would still allow 2 FX to be used.

arphex, 2021-10-10:
>> I know we can use one of the FX slots for tuning to get a similar result but this proposal would still allow 2 FX to be used.

we can do that through FX only for micro-tuning though right? which is not as flexible

darenager, 2021-10-10:
+/-24 semitones using U (tune) fx type as well, still hope the request will be considered.