polyend/TrackerBetaTesting #1154: [1.4.0b3] Lock up continuous tone

darenager, 2021-07-12:
## Tracker locked up and emitted a loud tone

### Steps to reproduce

I was setting up a LFO, I set a finetune LFO up, then without leaving the LFO page tried to go to set up a volume one, whilst changing the first column LFO target page, it just did the tone and locked up, I had to remove power then it was ok on rebooting, I was able to set the LFO.

* How often does this bug happen? It only happened once

* Does this bug happen in previous firmware versions? Not sure

* Does this bug happen in every project? Not sure

craighitchings, 2021-07-12:
This exact thing has happened to me too. I was using the stable 3.1 version at the time. I found that moving slowly through the mod pages helped, but there is definitely a bug here.

Jacknametrouble, 2021-07-13:
How were your Trackers powered while this happened? Does it happen when powered from the original AC adapter?

craighitchings, 2021-07-13:
`How were your Trackers powered while this happened? Does it happen when powered from the original AC adapter?`

It was a while ago for me, and as far as I remember, I was using Tracker standalone with no other equipment connected and being powered from the official Polyend adapter.

darenager, 2021-07-13:
Same here, using the PSU supplied. I also had the tone and lock up when removing then reinserting the SD card after a few minutes of it being out.
That does not always happen either though.

darenager, 2021-07-13:
Yes @craighitchings @Jacknametrouble looks like it is related to that issue.

closed 2021-12-18 by ambv

Jacknametrouble, 2022-01-15:
Fixed in firmware v1.5.0