polyend/TrackerBetaTesting #1162: [1.4.0b3] Inconsistent render selection

darenager, 2021-07-17:
When rendering a selection the results are incorrect and incosistent.

### Steps to reproduce

Highlight a selection you want to render, I used a whole track of 32 steps, went to top and used select all, shift+up.

Then render to selection, I noticed that not all steps had been rendered, repeat render to selection, again not all steps were rendered but different than last time with exact same selection, tried again and different again. After a few more attempts it rendered correctly, I continued repeating and the results were again inconsistent.

The easiest way to verify this bug is to do the first render, then go to the rendered file in the browser (exports/song name/selection) then after each render attempt go back to the sample loader and the new renders will be here to preview, I just used default names selection001 etc.

* How often does this bug happen? (every time, often, "it only happened once") Most of the time.

* Does this bug happen in previous firmware versions? Not sure

* Does this bug happen in every project? Yes, I tested a few and it happened on them all.

joeyfivecents, 2021-07-18:
This happens to me too. It takes 4 or 5 attempts to get the selection rendered correctly.

trutypesounds, 2021-09-26:
its the same on 1.4.0

ziej8asek, 2021-11-10:
Duplicate of #1251

closed 2021-11-10 by ziej8asek