polyend/TrackerBetaTesting #1164: Scream color or put more contrast

ketumatek, 2021-07-21:
Hi I have I big problem on summer I like to go outside to make music ...the problem is that on the screen we don t see so mucke the line and the number on the left...
It more easy to see the  note and the fx part than the other thing...
It is true that this machine is more use inside a studio or house but I m sure a lot of people use it outside...
May be be put more wite on the letter as you have on a tablette...
I will put a photo to see how it is...
Hope you can do something for this little problem...[image]

trutypesounds, 2021-07-21:
a different theme could solve this problem, or a way to customize it

ketumatek, 2021-07-21:
> a different theme could solve this problem, or a way to customize it

Yes very good idea ...not sure if they can ...but really cool

trutypesounds, 2021-07-21:
> > a different theme could solve this problem, or a way to customize it
> 
> Yes very good idea ...not sure if they can ...but really cool

Also removing the screen protection sticker could help, because its very glare, the display itself is matte which is a plus.

ketumatek, 2021-07-21:
> > > a different theme could solve this problem, or a way to customize it
> 
> > 
> 
> > Yes very good idea ...not sure if they can ...but really cool
> 
> 
> 
> Also removing the screen protection sticker could help, because its very glare, the display itself is matte which is a plus.

Yes it should be change a bit ...but not shire so much...
I let it on the screen because of more protection...I should be nice to put a glass protector as you can put on a phone...