polyend/TrackerBetaTesting #1170: Feature request - saving slices to their own instrument slots

Anexium, 2021-07-24:
I think the ability to save out the slices of a sample to their own instrument slots would be pretty handy.  For example, you'd take a break and slice it into 8.  You could then copy each of these slices to their own instrument.

Your instrument slots would then look like - 

01 - Break
02 - Break (Slice 1)
03 - Break (Slice 2)
…
09 - Break (Slice 8)

Admittedly it would double the amount of memory used, but the ability to be able to edit each slice and put different filters, EQ, loop-points, etc on would outweigh the increase in memory usage (you could always delete the original is space becomes an issue)

I know you can do this now using the resample function, but it's a bit of an unwieldy process.

tfspsound, 2021-07-27:
This would be a useful function since Performance FX such as Sample  Length don’t work on beat slices. It would also help us to quickly edit a sample that for whatever reason we were unable to edit at the time we recorded it.

Jacknametrouble, 2021-12-18:
This won't be possible in this exact form but since 1.5 beta1 it is possible to save the instruments with all their properties.

closed 2021-12-18 by Jacknametrouble