polyend/TrackerBetaTesting #1172: On mixer, summing, headroom, eq'ing, filtering

trutypesounds, 2021-07-26:
Here is just an observation on how the whole mixing/gain staging going on in the tracker.
Already read a lot of reports/complains on clipping, headroom issues here....

So, the deal is, when you mix different audio sources/samples there are of course frequencies that clash a lot, mostly the low ones which causes instant lack of headroom, clipping, muddiness, etc. So in the case of how PT is at the moment we have 3 filters (lo/hi pass & bandpass) and we have eq which isnt working at all. 

At first it seems you can just use the filters that already exist to hi pass the low freqs (so when summing the lows dont stack up like mad eating up all headroom, makes the track clipping and muddy) or/and lo pass to tame the highs if need, but in this case you are out of ability to modulate the filter then which is bad.

On offline-eq effect - it doesnt work atm, so when its gonna be fixed to work in a proper way it will help to sculpt the sound greatly of course, and yes you will be able to hipass the low frequencies, so it seems the problem would be solved but no. Even if the sample is hipassed in case of using different pitches (extreme ones for example) and filter modulations you again will have unwanted low frequencies going on around messing with the mix which revert us back to inability to sculpt a nice sounding mix.

So here is the question to the devs - is it possible to solve this problem somehow implementing some sort of additional locut/hicut eq and/or filter (6/12/24/666 slopes) in the mix chain? Just to affect the final sound on the channels, not the samples itself so its gonna solve the problem of headroom, muddines, clipping, etc.

This is a very important thing to have on device imo. Sadly i hadnt luck to fully work with such devices as Elektron’s Digitakt, Octatrack, but for example i’ve tested their Model:Cycles, yes its based on synthesis and probably they did a clever workaround for this issue, so the audio sum doesnt clip/gets mudded because of low freqs, always sounding loud and fat in a good way.

I also own OP-1 and had no issues with lack of headroom/low freqs clashing there with various type of samples/synth/tape combo and their drive fx on the master channel (which simply is a clever saturation compression fx) works very well.

Any thoughts, ideas, etc are very welcome @Jacknametrouble @piotr3500 @ambv @shuler323 

p.s. This also applies to the master reverb/delay busses, lo/hi passing is a must there.

Jacknametrouble, 2021-07-27:
The eq works, it is subtle but it works. The bass will go into distortion pretty fast because of the headroom. The solution to this is to have lower volume samples or just turn down the volume on instrument settings.

trutypesounds, 2021-07-27:
> The eq works, it is subtle but it works. The bass will go into distortion pretty fast because of the headroom. The solution to this is to have lower volume samples or just turn down the volume on instrument settings.

Eq is kinda "works", its so subtle like its nothing going on at all, atm its not possible to cut the lows properly, may be u need like x20 passes to cut 100hz lol

On headroom, hipassing the lows will leave you more headroom for loud an clear mixes, lowering the volume is not the solution, leaving you with quiet muddy mixes.

bndofc, 2021-07-27:
indeed, EQ does not work at all, i wouldn't even call it subtle, it straight up doesn't do anything if you don't repeat the process around 20 times like @trutypesounds mentioned.

tfspsound, 2021-07-27:
I think that having some form of Q control would help particularly with the band pass filter. If we could control the width of the band of frequencies effected by the filter there would be less over lapping of similar frequencies. Maybe the ability to set upper and lower cutoff frequency’s would help too. Not sure where they would fit on screen though. Currently it feels like the louder sample sounds and lesser samples get lost or add to the muddy sound particularly when using the limiter to create a side chain effect (which sounds great btw) It becomes a bit of a balancing act of taking a bit off the track volume here and adding a bit to the sample volume there, which can get thrown out completely with addition of a new sample or section. A notch filter could be helpful too for eliminating specific frequencies, if possible.

theorize999, 2021-07-31:
People make it work in FT2 with absolutely no filters or FX at all. No offense but it sounds like you're trying to force samples that don't work for what you're trying to do with them. PT is pretty upfront about it's intention to set limits, And I think we've also forgotten that we can resample, so as far as editing samples with a high pass that functionality is already there, not that the eq in the sample editor doesn't need work, but I think there should just be a high/lowpass filter in the sample editor, I've always thought they should put that in. 
I'm all for a variable width BP, just in addition to the one we already have and I think it should be non-resonant. I'd also be down with a BP like the one we already have but with a steeper slope, to me it seems like there's not much wiggle room with the BP for modulating it. 
I'm all for having a separate hp in the instrument, but I'd rather see it be an option, like switch out the drive for HP kind of thing. I just don't think we should blame the tool for how we're using it. Personally, I think a lot of crap in PT that if it wasn't there maybe we could have more tracks. I'll never ever use the live mode, I would love to have the resources it eats up for actual tracking. But it's a hardware tracker who am I to complain, and I've found it really starts to shine when you use midi synths for these technical things and use the sampler for the rest, PT plus blofeld is a badass combo. Hope I didn't offend anyone because I mean all this with care and respect, ask for improvements, just don't blame the tool for your mixes because people have gotten along for a long time with far less, and PT is exactly what it is and is pretty damn upfront about it.

trutypesounds, 2021-07-31:
> People make it work in FT2 with absolutely no filters or FX at all. No offense but it sounds like you're trying to force samples that don't work for what you're trying to do with them. PT is pretty [...]

well, i kinda agree on FT2 statement, but FT2 sound engine is a very different beast and it worked in close relation to the soundblaster/gravis hardware and it sounded different. also may be all of us forgot how NON bass all the sound from FT2 was, on par with IT (which had filters btw!).
of course you can prepare all your samples on computer (or resample on PT with the filters on it) but i want to rely more on PT itself, radio sampling, external stuff sampling, so eq/filter sculpting is essential i guess. i mean come on, every single cheap ass hardware mixer have a locut button.

All of this are of course for enhancements and ease of workflows and not for blaming the devs or something, im still having WOW and WTF moments (in a good way) after 1,5 months of working with PT, its a great device with its own quirks and stuff, thank god finally somebody built such a device, it's fire!

theorize999, 2021-08-01:
I am totally with you on the self sufficient PT thing, I hope they continue to work on the editor and make resampling smoother, I actually picked up renoise a couple months ago for doing more in depth stuff, I love my hardware rig, but also love Tone2 VST's!!!! Plus sometimes I want to be able to do whatever I can think of. And yes FT2 seems to have a different bit of a different sound to it, I saw online a long time ago that it has a built in limiter I'm guessing it's doing a lot of that. I've been loosely keeping up with the 8bb clone I need to spend more time with it now that it has higher quality audio and interpolation options, I'm really interested in that. I actually agree with you, I think it is hard to get results like I would get in other environments (even FT2, quadruple the tracks makes a huge difference!), I've gotten sound that I really like, but it's just different because of the choices I'm forced to make. I agree because with the low track count you have to make your sounds bigger... to get a full sounding mix it can get a little messy... it really would be nice to have some extra filtering. I think it's getting there but like I eluded to I have it hooked up with other gear till it really is ready to compose fully self contained. If they would do a good offline noise cleaner like using the selection as a reference thing it would already be one hell of a sound design station, like for movies/games/websites type. I mean it already can be but to be able to clean audio would make it so much more versatile for sound design. IDK I could go on for days about all the changes I would do (drum track would be high on that list), I hope they get a lot of it in there once it's in a true self contained state I'll buy another one maybe two more. 24 tracks of PT would be very nice.  My bad I tend to go on and I didn't mean to sound like I was judging I was just tired I've only had about 4 hours free this week, super busy. I totally agree but I do like the whole limitation thing and the challenge of it, but certain things do need to be in it for it to be truly self contained.

TheGrizard, 2022-09-01:
Can we get a scale on the meters in the track mixer/mix pages? I tend to drop the faders on the track mixer page and don’t have issues with headroom - be nice to see what level the audio peaks are hitting so we can build a mix visually (as well as by ears!)