polyend/TrackerBetaTesting #1173: Button press continuity inconsistency

trutypesounds, 2021-07-26:
I'm not sure if i named this correctly, but ill try to explain whats the deal here, its a minor issue, but could be great if fixed.

When on various screens (for example sample playback or sample editor or config) there is buttons doubled as up/down, if i use arrow keys on the right side of the tracker and press the button the cursor will go up and down continuously with little acceleration, but if i used buttons under the screen the cursor just move once, which could be pretty annoying, in sample editor for example, where you must press the buttons many time to constantly moving between crop/normalizer/timestretch/fadeinout. Yes i can use jog to select fast but prefer to use buttons to minimize jog usage so it doesnt wear fast.

See video below

https://user-images.githubusercontent.com/17484006/127036076-2cda7c22-da2e-4bf3-8c02-48c385c7a8b1.MOV

trutypesounds, 2021-12-25:
any news on this one? @ambv @Jacknametrouble @piotr3500 @wjakobczyk-polyend 
in song mode for example if i suddenly want to add a new pattern in already arranged song, lets say its pattern 80, i need to manually press button 80 times, instead of just press and hold and wait till the number will reach 80

Jeymadcat, 2022-01-18:
Yes its the same ! like that ! exactly !! thanks for remak the video : (   please please ! resolve this  f*** bug

bambrose, 2022-01-18:
Yes ! This is a great explanation
you’ve identified a real annoyance that was hard to work out but when pointed it is big deal
It’s REALLY evident in sample editor

trutypesounds, 2022-01-18:
> Yes ! This is a great explanation you’ve identified a real annoyance that was hard to work out but when pointed it is big deal It’s REALLY evident in sample editor

@Jacknametrouble @MortalJS @ambv @piotr3500 @wjakobczyk-polyend its obviously much more important thing in terms of usability than just renaming Step to Step jump (even why renaming it?)

Jacknametrouble, 2022-01-18:
@trutypesounds please do not tag everyone you can, we read the posts. 
Step function was renamed to Step Jump because it was confusing for many users.
Especially that we also call a step consisting of a note, instrument and fx's a step.
Hope this clears the topic.

trutypesounds, 2022-01-19:
> @trutypesounds please do not tag everyone you can, we read the posts. Step function was renamed to Step Jump because it was confusing for many users. Especially that we also call a step consisting of a note, instrument and fx's a step. Hope this clears the topic.

thanks for the explanations, also hope this enhancement i've mentioned will be implemented in next firmware iterations, it would save those func buttons under screen and jog from a lot of presses and rotations (also mental state of users, nobody wants to press the button 80 times to do something)