polyend/TrackerBetaTesting #1174: [1.3.1] EQ doesn’t work on new recording

frognerboy, 2021-07-27:
I recorded a voice snippet with an SM57, which played back fine. When trying to apply EQ in sample editor nothing happens. All other effects seem to work.

* How often does this bug happen? (every time, often, "it only happened once") Every time

* Does this bug happen in previous firmware versions? N/a

* Does this bug happen in every project? N/a

trutypesounds, 2021-07-27:
> I recorded a voice snippet with an SM57, which played back fine. When trying to apply EQ in sample editor nothing happens. All other effects seem to work.
>
> * How often does this bug happen? (every [...]

Mentioned many times here, yes its not working.

MortalJS, 2022-01-17:
Hi @frognerboy,

Thanks for your feedback.

I checked your post on the current version 1.5.0 and 1.3.0.

The performance for high frequencies has been improved in 1.5.0.

As you are able to see its maximum ranges are +/- 12 db and within this range the editing occurs.
If you want larger values apply the effect several times.

I am closing the thread.

closed 2022-01-17 by MortalJS

trutypesounds, 2022-01-18:
@MortalJS its not fixed, eq still works not like eq guys, please make it the right way!